SiliconLabs/matter #34: activate.sh script fails on Raspberry Pi

below, 2022-09-16:
#### activate.sh script fails on Raspberry Pi
The `activate.sh` or `bootstrap.sh` scripts fail on a Raspberry Pi 4 

### Expected behavior
Scripts fail without error
### actual behavior
An Error occurs: `× Building wheel for cryptography (pyproject.toml) did not run successfully.`
### system configuration
```Linux MatterPi 5.15.0-1014-raspi #16-Ubuntu SMP PREEMPT Tue Aug 23 10:22:52 UTC 2022 armv7l armv7l armv7l GNU/Linux```
### Other Info
```
$ pip install wheel
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: wheel in /usr/lib/python3/dist-packages (0.37.1)
```
```
$ pip install cryptography
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: cryptography in /usr/lib/python3/dist-packages (3.4.8)
```
### Full Log:
[fulllog.txt](https://github.com/SiliconLabs/matter/files/9582473/fulllog.txt)

Any help is appreciated …

silabs-Simran, 2022-09-20:
You can follow the pre-requisite guide to setup Raspi 4. This guide is old but you need to install library files mentioned in this article: https://community.silabs.com/s/article/setup-rpi4-matter-development?language=en_US

silabs-Simran, 2022-10-17:
Closing it.

silabs-Simran, 2022-10-17:
Closing it.

closed 2022-10-17 by silabs-Simran