SiliconLabs/matter #37: problem with software artifact for OTBR - incorrect password

timredfern, 2022-09-19:
I'm trying to follow the documentation published from the workwith conference workshop MAT-301 on setting up the openthread border router using the filesystem image published on the software artifact page here. The adress of the FS image is https://www.silabs.com/documents/public/software/SilabsMatterPi.zip

The image can be flashed but the login supplied in "MAT-301 Lab Manual.docx" does not work.

I know I am sshing to the correct address as the challenge uses EDD25519 as per the intstructions. Also the device can be powered off and the server does not respond.

```
tim@MacBook-Pro-4 ~ % ssh ssh ubuntu@192.168.15.51
ssh: Could not resolve hostname ssh: nodename nor servname provided, or not known
tim@MacBook-Pro-4 ~ % ssh ubuntu@192.168.15.42                                                            
ssh: connect to host 192.168.15.42 port 22: Connection refused
tim@MacBook-Pro-4 ~ % ssh ubuntu@192.168.15.51
The authenticity of host '192.168.15.51 (192.168.15.51)' can't be established.
ED25519 key fingerprint is SHA256:+YYu+ycaTtzNojj9aQ7agmhDhFr4Qv447GRIqYK5H8A.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.15.51' (ED25519) to the list of known hosts.
ubuntu@192.168.15.51's password: 
```
The login supplied in the doc is `raspberrypi` however this fails:

```
ubuntu@192.168.15.51's password: 
Permission denied, please try again.
```

timredfern, 2022-09-19:
it turns out that the password is still the default (ubuntu)

silabs-Simran, 2022-09-19:
Please refer to the documentation on [Raspberry Pi Login Credentials](https://github.com/SiliconLabs/matter/blob/release_0.3.0/docs/silabs/thread/RASPI_IMG.md#raspberry-pi-login-credentials)

closed 2022-09-19 by silabs-Simran