SiliconLabs/matter #43: interop_testing_te9 branch not working because of missing thirdy_party/telink_sdk

q7777777hk, 2022-10-05:
interop_testing_te9 branch not working because of missing thirdy_party/telink_sdk

Can fix the link ?

silabs-sebastien, 2022-10-14:
@q7777777hk Are you referring to this link? https://github.com/SiliconLabs/matter/tree/release_0.4.0/third_party/telink_sdk

Please indicate the steps you followed to initialize your local repo and any other steps required to reproduce this issue.

q7777777hk, 2022-10-14:
When you check the telink_sdk in https://github.com/SiliconLabs/matter/tree/interop_testing_te9/third_party/telink_sdk, 
you will see 404 

After "git clone -b interop_testing_te9 --recursive https://github.com/SiliconLabs/matter.git"
Run "git submodule update --init --recursive" will cause fail
Then cannot run "source third_party/connectedhomeip/scripts/activate.sh"
And cannot build any project

silabs-sebastien, 2022-10-25:
Please note that the TE9 branch is quite old and not up to date at all. Is there a specific reason why you are trying to use this older branch vs the newest release (i.e.: [0.4.0](https://github.com/SiliconLabs/matter/releases/tag/v0.4.0))?

silabs-sebastien, 2022-10-28:
Considering the TE9 branch will not be supported going forward and most likely removed once Matter 1.0 is released, I recommend trying out the latest release, 0.4.0. If you need further assistance, please open a ticket on the community [here](https://community.silabs.com/s/topic/0TO1M000000qHZgWAM/matter?language=en_US).

closed 2022-10-28 by silabs-sebastien