SiliconLabs/matter #52: Thread demo doesn't work, documentation is incomplete

timredfern, 2022-10-21:
### Description of the issue

I tried to follow the instructions https://github.com/SiliconLabs/matter/blob/release_0.4.0/docs/silabs/thread/DEMO_OVERVIEW.md

The prebuilt OTBR image for RPi doesn't work as described. The `mattertool startThread` command fails  with `connect session failed: No such file or directory`

Digging a bit deeper, I followed the instructions here: https://openthread.io/guides/border-router/build

The MG24 RCP does work, it's necessary to use the OTBR web gui to start the thread network before the matter device can join the network. The instructions here are just incomplete and don't include any troubleshooting.

### Steps to reproduce the issue

```bash
try to follow https://github.com/SiliconLabs/matter/blob/release_0.4.0/docs/silabs/thread/DEMO_OVERVIEW.md
```

### Device type(s) and network topology

RPi4 with PCB2703 dev kit connected via USB

### Proposed solution(s)

fix the demo instructions or just direct user to follow openthread tutorial

### SMG version

release_0.4.0

### Protocol(s)

Openthread

### Hardware platform(s)

brd2703a

silabs-sebastien, 2022-10-24:
I recommend you give a try to [this guide](https://community.silabs.com/s/article/MAT-301-Lab-Update-Works-With-2022?language=en_US) and make sure to check all the steps. This guide mentions release 0.3.0 so do make sure to use 0.4.0 links instead.

If you still hit any issues, please open a support ticket in the [Silabs Matter Community](https://community.silabs.com/s/topic/0TO1M000000qHZgWAM/matter?language=en_US).

closed 2022-10-24 by silabs-sebastien

timredfern, 2022-10-25:
I don't actually see a step by step guide at your link, it's a set of corrections to the Workswith workshop instructions?

I have got OTBR working from google's instructions. The documentation in this repo is misleading though.

silabs-sebastien, 2022-10-25:
The link above includes a link to: _Comprehensive [README](https://github.com/SiliconLabs/matter/blob/v0.3.0/README.md) for instructions to run our examples_
This [README](https://github.com/SiliconLabs/matter/blob/release_0.4.0/README.md) links to [full guide](https://github.com/SiliconLabs/matter/blob/release_0.4.0/docs/silabs/README.md). I've copied here the 0.4.0 links instead of 0.3.0 for convenience.

Again, if you still have any issues, please open a support ticket in the [Silabs Matter Community](https://community.silabs.com/s/topic/0TO1M000000qHZgWAM/matter?language=en_US).