SiliconLabs/matter #58: Matter Attestation Credentials for EFR32 failed

cjszz2001, 2022-11-01:
### Description of the issue

Did Matter Attestation Credentials for EFR32 example successfully on the development through brd4186c by following this instructions. https://github.com/SiliconLabs/matter/blob/release_0.3.0/silabs_examples/credentials/README.md
But when we use the same procedures to download to the contact sensor board which has the EFR32MG24B010F1536IM48 chip. It failed at the commissioning stage.

### Steps to reproduce the issue

```bash
Device Setup
Enter the creds.py directory:
cd silabs_examples/credentials

Set the environment:
export BASE_SDK_PATH=<Gecko SDK path>
export ARM_GCC_DIR=<ARM GNU toolchain path>

Example:

export BASE_SDK_PATH=/home/myUserName/matter/third_party/silabs/gecko_sdk/
export ARM_GCC_DIR=/usr

Execute the setup script
python3 ./creds.py -p /dev/ttyACM0 -S 440266330 -B brd4186c -C pai_cert.pem -K pai_priv.pem -D cd.bin
```

### Device type(s) and network topology

Google home apps through Nest Hub over thread network with matter contact sensor which has the EFR32MG24B010F1536IM48 chip.

### SMG version

release_0.1.0

### Protocol(s)

Openthread

### Hardware platform(s)

 EFR32MG24B010F1536IM48 

### Development platform(s)

 contact sensor board which has the EFR32MG24B010F1536IM48 chip to detect gpio open/close

### Captured data or logs

[failed_commisonsion_on_contact_sensor.txt](https://github.com/SiliconLabs/matter/files/9913258/failed_commisonsion_on_contact_sensor.txt)
[success_commsion_on_dev_board.txt](https://github.com/SiliconLabs/matter/files/9913259/success_commsion_on_dev_board.txt)
[excute_setup_script_on_contact_board.txt](https://github.com/SiliconLabs/matter/files/9913260/excute_setup_script_on_contact_board.txt)

cjszz2001, 2022-11-04:
I do not have this build issue on release_0.3.0.

silabs-sebastien, 2022-11-07:
Did you also test this on 0.4.0 or 1.0.0? If not, I highly recommend trying it out on 1.0.0 and check if you get a different result.

cjszz2001, 2022-11-07:
I just test on 1.0.0. It works on the development board. But it failed on our contact sensor board.
It gets stuck at the requesting CSR from the contact sensor board.

Here is the log:
python3 ./creds.py -p /dev/ttyACM0 -S 440208289 -B brd4186c -N "Matter dev" -V 4144 -P 37635 -C pai_cert.pem -K pai_priv.pem -D cd.bin
 
Building device app
  make -C device/brd4186c -f device-creds.Makefile

make: Entering directory '/home/builder/matter/silabs_examples/credentials/device/brd4186c'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/builder/matter/silabs_examples/credentials/device/brd4186c'

Flashing device app
  commander flash device/brd4186c/build/debug/device-creds.s37 --serialno 440208289

Parsing file device/brd4186c/build/debug/device-creds.s37...
Writing 65536 bytes starting at address 0x08000000
Comparing range 0x08000000 - 0x0800FFFF (64 KB)
DONE

Building host app
  make -C host/app/ -f host-creds.Makefile

make: Entering directory '/home/builder/matter/silabs_examples/credentials/host/app'
Linking build/debug/host-creds
Done.
make: Leaving directory '/home/builder/matter/silabs_examples/credentials/host/app'

Creating temp dir
  mkdir -p ./temp/

Requesting CSR
  ./host/app/build/debug/host-creds -p /dev/ttyACM0 -f ./temp/csr.pem -N Matter dev -V 4144 -P 37635

Requesting CSR (cn:'Matter dev',vid:0x1030, pid:0x9303) from '/dev/ttyACM0' as './temp/csr.pem'
Requesing CSR...

cjszz2001, 2022-11-08:
I added some debug log code in the host code, it indicates serial port can not open.
I am just wondering it is the Vcom port which is enabled on the radio board. But our sensor contact does not have the Vcom port.

cjszz2001, 2022-11-15:
I edited host code to set serial port speed to 115200. now it can do CSR request through ttyUSB0. 
Through ttyACM0, it will do flashing through dev board out mode to the sensor.
Now it could do the setup scripts successfully on the sensor. But still failed at the commission. It works on the 4186c development board.
please check the log.

[excute_setup_script_on_contact_board.txt](https://github.com/SiliconLabs/matter/files/10016651/excute_setup_script_on_contact_board.txt)
[failed_commisonsion_on_contact_sensor.txt](https://github.com/SiliconLabs/matter/files/10016653/failed_commisonsion_on_contact_sensor.txt)

silabs-sebastien, 2023-01-11:
From the logs you provided, it seems the certificate injection was OK. That being said, if it fails at the commissioning stage it is likely the VID & PID do not match the information contained on the CD (certification declaration) or the VID doesn't match the PID (intermediate certificate) & PAA (root certificate).

Do you have more details since your last post about this? Did you have other results?

closed 2023-01-27 by github-actions[bot]