SiliconLabs/matter #66: I'm not able to build an image with TCP enabled

lecontr2, 2022-11-16:
### Description of the issue

Hi, I've tried enabling the TCP options in the OpenThreadConfig.h header file to test the TCP client application, but my image fails to build. It appears that a linking issue occurs in respect to the function definitions in tcp_api.cpp are missing from the final build. I've tried working backwards by checking the function definitions for the udp_api, but I haven't been able to figure out why those are included but tcp isn't. Is there something I'm missing?

### Steps to reproduce the issue

```bash
1. Enable OPENTHREAD_CONFIG_TCP_ENABLE in OpenThreadConfig.h in lighting application.
2. Run build script.
```

### Device type(s) and network topology

N/A
Note: I'm using release_1.0.0

### SMG version

release_0.4.0

### Protocol(s)

Openthread

### Hardware platform(s)

MG24 Explorer Kit, BRD4186C

### Captured data or logs

[20/27] arm-none-eabi-g++ -T../../../examples/lighting-app/efr32/third_party/connectedhomeip/examples/platform/efr32/ldscripts/efr32mg24.ld -march=armv8-m.main+dsp -mcpu=cortex-m33 -mabi=aapcs -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -Og --specs=nosys.specs --specs=nano.specs -Werror -Wl,--fatal-warnings -fdiagnostics-color -Wl,--gc-sections -Wl,--wrap=malloc -Wl,--wrap=free -Wl,--wrap=realloc -Wl,--wrap=calloc -Wl,--wrap=MemoryAlloc -Wl,--wrap=_malloc_r -Wl,--wrap=_realloc_r -Wl,--wrap=_free_r -Wl,--wrap=_calloc_r -Wl,-Map,./chip-efr32-lighting-example.out.map @./chip-efr32-lighting-example.out.rsp -o ./chip-efr32-lighting-example.out
FAILED: chip-efr32-lighting-example.out chip-efr32-lighting-example.out.map 
arm-none-eabi-g++ -T../../../examples/lighting-app/efr32/third_party/connectedhomeip/examples/platform/efr32/ldscripts/efr32mg24.ld -march=armv8-m.main+dsp -mcpu=cortex-m33 -mabi=aapcs -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -Og --specs=nosys.specs --specs=nano.specs -Werror -Wl,--fatal-warnings -fdiagnostics-color -Wl,--gc-sections -Wl,--wrap=malloc -Wl,--wrap=free -Wl,--wrap=realloc -Wl,--wrap=calloc -Wl,--wrap=MemoryAlloc -Wl,--wrap=_malloc_r -Wl,--wrap=_realloc_r -Wl,--wrap=_free_r -Wl,--wrap=_calloc_r -Wl,-Map,./chip-efr32-lighting-example.out.map @./chip-efr32-lighting-example.out.rsp -o ./chip-efr32-lighting-example.out
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: obj/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/ot-efr32-cert.cli_tcp.cpp.o: in function `ot::Cli::TcpExample::ProcessInit(ot::Utils::CmdLineParser::Arg*)':
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/out/lighting-app/BRD4186C/../../../examples/lighting-app/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/cli_tcp.cpp:111: undefined reference to `otTcpEndpointInitialize'
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /media/lecontr1/Extension/workspace/silabs-v.1.0/matter/out/lighting-app/BRD4186C/../../../examples/lighting-app/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/cli_tcp.cpp:122: undefined reference to `otTcpListenerInitialize'
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /media/lecontr1/Extension/workspace/silabs-v.1.0/matter/out/lighting-app/BRD4186C/../../../examples/lighting-app/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/cli_tcp.cpp:125: undefined reference to `otTcpEndpointDeinitialize'
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: obj/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/ot-efr32-cert.cli_tcp.cpp.o: in function `ot::Cli::TcpExample::ProcessDeinit(ot::Utils::CmdLineParser::Arg*)':
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/out/lighting-app/BRD4186C/../../../examples/lighting-app/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/cli_tcp.cpp:145: undefined reference to `otTcpEndpointDeinitialize'
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /media/lecontr1/Extension/workspace/silabs-v.1.0/matter/out/lighting-app/BRD4186C/../../../examples/lighting-app/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/cli_tcp.cpp:148: undefined reference to `otTcpListenerDeinitialize'
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: obj/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/ot-efr32-cert.cli_tcp.cpp.o: in function `ot::Cli::TcpExample::ProcessBind(ot::Utils::CmdLineParser::Arg*)':
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/out/lighting-app/BRD4186C/../../../examples/lighting-app/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/cli_tcp.cpp:169: undefined reference to `otTcpBind'
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: obj/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/ot-efr32-cert.cli_tcp.cpp.o: in function `ot::Cli::TcpExample::ProcessConnect(ot::Utils::CmdLineParser::Arg*)':
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/out/lighting-app/BRD4186C/../../../examples/lighting-app/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/cli_tcp.cpp:186: undefined reference to `otTcpConnect'
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: obj/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/ot-efr32-cert.cli_tcp.cpp.o: in function `ot::Cli::TcpExample::ProcessSend(ot::Utils::CmdLineParser::Arg*)':
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/out/lighting-app/BRD4186C/../../../examples/lighting-app/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/cli_tcp.cpp:208: undefined reference to `otTcpSendByReference'
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: obj/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/ot-efr32-cert.cli_tcp.cpp.o: in function `ot::Cli::TcpExample::ProcessBenchmark(ot::Utils::CmdLineParser::Arg*)':
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/out/lighting-app/BRD4186C/../../../examples/lighting-app/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/cli_tcp.cpp:248: undefined reference to `otTcpSendByReference'
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: obj/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/ot-efr32-cert.cli_tcp.cpp.o: in function `ot::Cli::TcpExample::ProcessSendEnd(ot::Utils::CmdLineParser::Arg*)':
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/out/lighting-app/BRD4186C/../../../examples/lighting-app/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/cli_tcp.cpp:268: undefined reference to `otTcpSendEndOfStream'
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: obj/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/ot-efr32-cert.cli_tcp.cpp.o: in function `ot::Cli::TcpExample::ProcessAbort(ot::Utils::CmdLineParser::Arg*)':
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/out/lighting-app/BRD4186C/../../../examples/lighting-app/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/cli_tcp.cpp:281: undefined reference to `otTcpAbort'
[... cut ...] /src/cli/ot-efr32-cert.cli_tcp.cpp.o: in function `ot::Cli::TcpExample::HandleTcpAcceptDoneCallback(otTcpListener*, otTcpEndpoint*, otSockAddr const*)':
/media/lecontr1/Extension/workspace/silabs-v.1.0/matter/out/lighting-app/BRD4186C/../../../examples/lighting-app/efr32/third_party/connectedhomeip/third_party/openthread/ot-efr32/openthread/src/cli/cli_tcp.cpp:371: undefined reference to `otTcpListenerGetContext'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

lecontr2, 2022-11-18:
Is there any way we can update the static libraries from OpenThread to include TCP? It seems like they only include the UDP library at the moment.

silabs-sebastien, 2022-12-07:
Hi @lecontr2 ,

Please note that TCP in Matter over Thread is currently not supported, hence why it is unavailable when you try to compile it into your project.

You can see this example [here](https://github.com/project-chip/connectedhomeip/blob/master/src/inet/TCPEndPointImplOpenThread.cpp), where a stub API is all that exists.

Sincerely,

closed 2022-12-07 by silabs-sebastien