SiliconLabs/matter #67: BRD4161A Prebuilt image doesn't work

HiFiPhile, 2022-11-17:
### Description of the issue

BRD4161A Rev A03 board flashed with `matter_accessory_device_binaries_1.0.0.zip\matter_accessory_device_binaries_1.0.0\light_switch_app_thread\OpenThread\release\BRD4161A\chip-efr32-light-switch-example.s37`:

- LED0 blink once every 1s, looks like watchdog reset.
- No output from serial console
- No output on devkit LCD

### Steps to reproduce the issue

```bash
1. Download matter_accessory_device_binaries_1.0.0.zip
2. Flash light_switch_app_thread\OpenThread\release\BRD4161A\chip-efr32-light-switch-example.s37
```

### Device type(s) and network topology

BRD4161A devkit

### SMG version

release_0.4.0

### Protocol(s)

Openthread

### Hardware platform(s)

BRD4161A

### Development platform(s)

Windows 10

silabs-sebastien, 2022-11-29:
Hi @HiFiPhile,

I've tried a BRD4161A with the binary you mentioned (4161A light-switch) and the 4161A combined bootloader. Unlike you, I got LCD output, but I also do get the same LED0 blinking ~1/s pattern you've mentioned and no output on the serial.

If you press BTN1, do you get any changes on LED1?

HiFiPhile, 2022-11-30:
@silabs-sebastien 
> I got LCD output

What did you got on LCD ? I wasn't very clear, LCD always displays default J-Link content with Silabs logo, it's not controlled by the MCU.

> If you press BTN1, do you get any changes on LED1?

No, LED1 is always off.

silabs-sebastien, 2022-12-07:
@HiFiPhile 
Concerning what you reported for the binaries, it's actually quite sensible since you used the **release** binaries, who are geared toward saving footprint/byte size, The sleepy binaries are an easy guess: they focus on assisting in reducing power consumption for battery-powered devices.

Therefore, please try again using the **standard** binaries and see if you get the LCD screen and console outputs you expected.

HiFiPhile, 2022-12-08:
> Therefore, please try again using the standard binaries and see if you get the LCD screen and console outputs you expected.

Indeed I got QRCode output on LCD and Thread CLI on UART.

But the application crashed as soon as BLE commissioning is started, below is RTT log:
```
00> <efr32 > Factory Reset Triggered. Release button within 3000ms to cancel.
00> <info  > [TS] Pending Last Known Good Time: 2022-10-31T20:48:25
00> <info  > [TS] Previous Last Known Good Time: 2022-10-31T20:48:25
00> <info  > [TS] Reverted Last Known Good Time to previous value
00> <detail> [ZCL] Emitting ShutDown event
00> <detail> [EVL] LogEvent event number: 0x0000000000020002 priority: 2, endpoint id:  0x0 cluster id: 0x0000_0028 event id: 0x1 Epoch timestamp: 0x000000DC6AD01A0D
00> <detail> [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
00> <info  > [DL] Performing factory reset
00> <info  > [DL] Clearing Thread provision
00> <info  > [DL] Erasing Thread persistent info...
00> <info  > [DL] System restarting
00> <efr32 > ==================================================
00> <efr32 > SiLabs-Light-Switch starting
00> <efr32 > ==================================================
00> <efr32 > Init CHIP Stack
00> <efr32 > Initializing OpenThread stack
00> <info  > [DL] OpenThread started: OK
00> <info  > [DL] Setting OpenThread device type to ROUTER
00> <efr32 > Starting OpenThread task
00> <info  > [SVR] Server initializing...
00> <detail> [FP] Initializing FabricTable from persistent storage
00> <info  > [TS] Last Known Good Time: [unknown]
00> <info  > [TS] Setting Last Known Good Time to firmware build time 2022-10-31T20:46:44
00> <info  > [DMG] AccessControl: initializing
00> <info  > [DMG] Examples::AccessControlDelegate::Init
00> <info  > [DMG] AccessControl: setting
00> <info  > [DMG] DefaultAclStorage: initializing
00> <info  > [DMG] DefaultAclStorage: 0 entries loaded
00> <detail> [IN] UDP::Init bind&listen port=5540
00> <detail> [IN] UDP::Init bound to port=5540
00> <detail> [IN] BLEBase::Init - setting/overriding transport
00> <detail> [IN] TransportMgr initialized
00> <detail> [DL] Using Thread extended MAC for hostname.
00> <info  > [ZCL] Using ZAP configuration...
00> <detail> [DMG] Failed to read stored attribute (0, 0x0000_0028, 0x0000_0005: a0
00> <detail> [DMG] Failed to read stored attribute (0, 0x0000_0028, 0x0000_0010: a0
00> <detail> [DMG] Failed to read stored attribute (0, 0x0000_002C, 0x0000_0000: a0
00> <detail> [DMG] Failed to read stored attribute (0, 0x0000_002C, 0x0000_0001: a0
00> <info  > [DMG] AccessControlCluster: initializing
00> <detail> [DMG] Endpoint 0, Cluster 0x0000_0030 update version to 4fb9e7be
00> <info  > [ZCL] Cluster callback: 0x0000_0030
00> <detail> [ZCL] GeneralDiagnostics: OnDeviceReboot
00> <detail> [DMG] Endpoint 0, Cluster 0x0000_0033 update version to 889023fe
00> <detail> [EVL] LogEvent event number: 0x0000000000000000 priority: 2, endpoint id:  0x0 cluster id: 0x0000_0033 event id: 0x3 Epoch timestamp: 0x000000DC6ACFAC00
00> <info  > [ZCL] Initiating Admin Commissioning cluster.
00> <error > [ZCL] DeviceInfoProvider is not registered
00> <error > [ZCL] Trying to write invalid Calendar Type
00> <error > [ZCL] Failed to write calendar type with error: 0x87
00> <detail> [DMG] Endpoint 1, Cluster 0x0000_0003 update version to f522b51b
00> <info  > [ZCL] Cluster callback: 0x0000_0003
00> <info  > [ZCL] Identify attribute ID: 0x0000_0001 Type: 48 Value: 2, length 1
00> <detail> [DMG] Endpoint 1, Cluster 0x0000_0004 update version to 85702a66
00> <info  > [ZCL] Cluster callback: 0x0000_0004
00> <detail> [DMG] <error > [DIS] Failed to advertise commissionable node: 3
00> <error > [DIS] Failed to finalize service update: 1c
00> <info  > [DIS] Updating services using commissioning mode 1
00> <detail> [DL] Using Thread extended MAC for hostname.
00> <detail> [DL] Using Thread extended MAC for hostname.
00> <info  > [DIS] Advertise commission parameter vendorID=65521 productID=32772 discriminator=3840/15 cm=1
00> <error > [DIS] Failed to advertise commissionable node: 3
00> <error > [DIS] Failed to finalize service update: 1c
00> <info  > [DIS] Delaying proxy of operational discovery: missing dele<efr32 > Starting App Task
00> <efr32 > Starting FreeRTOS scheduler
00> <info  > [DL] Bluetooth stack booted: v4.0.0-b191
00> <info  > [DL] RAIL version:, v2.13.0-b2
00> <detail> [DL] Thread task running
00> <detail> [DL] CHIP task running
00> <detail> [DL] Start BLE advertissement
00> <efr32 > Current Software Version: 0.1ALPHA
00> <info  > [DL] Device Configuration:
00> <info  > [DL]   Serial Number: TEST_SN
00> <info  > [DL]   Vendor Id: 65521 (0xFFF1)
00> <info  > [DL]   Product Id: 32772 (0x8004)
00> <info  > [DL]   Hardware Version: 0
00> <info  > [DL]   Setup Pin Code (0 for UNKNOWN/ERROR): 0
00> <info  > [DL]   Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 3840 (0xF00)
00> <info  > [DL]   Manufacturing Date: (not set)
00> <info  > [DL]   Device Type: 65535 (0xFFFF)
00> <info  > [DL] _OnPlatformEvent default:  event->Type = 32777
00> <detail> [DL] OpenThread State Changed (Flags: 0x00038200)
00> <detail> [DL]    Network Name: OpenThread
00> <detail> [DL]    PAN Id: 0xFFFF
00> <detail> [DL]    Extended PAN Id: 0xDEAD00BEEF00CAFE
00> <detail> [DL]    Channel: 11
00> <detail> [DL]    Mesh Prefix: fdde:ad00:beef:0:0:0:0:0/64
00> <info  > [SVR] Cannot load binding table: a0
00> <efr32 > App Task started
00> <detail> [DL] bleAdv Timeout : Start slow advertissment
00> <info  > [DL] Connect Event for handle : 1
00> <efr32 > TASK OVERFLOW
00> <efr32 > BLE_STACK
00> 
00> FREERTOS ASSERT ( ( portNVIC_INT_CTRL_REG & portVECTACTIVE_MASK ) == 0 )
```