SiliconLabs/matter #70: silabs_examples/sl-newLight build failed

bytespectrumllp, 2022-11-24:
### Description of the issue

While builing the sl-newLight example for board BRD2601B
build failed with following error.
[image]

System: Ubuntu 20.04.4 LTS: 8GB RAM 64 bit core i5 7th gen.

Attaching the log file as well.

### Steps to reproduce the issue

```bash
1. clone Release 1.0.0
2. cd matter
3. export SILABS_BOARD=BRD2601B
4. Also tried with > export EFR32_BOARD=BRD2601B
5. source scripts/activate.sh
6. ./scripts/examples/gn_efr32_example.sh ./silabs_examples/sl-newLight/efr32 ./out/sl-newLight
```

### Device type(s) and network topology

EFR32MG24: Thunderboard sense devkit BRD2601B Matter end device

### SMG version

release_0.4.0

### Protocol(s)

Openthread, Wifi, Bluetooth

### Hardware platform(s)

xG24 Devkit Thunderboard sense: BRD2601B

### Captured data or logs

[err.log](https://github.com/SiliconLabs/matter/files/10084677/err.log)

[image]

grobasoz, 2022-12-17:
So it seems the ports and pins are hardwired in _led_widget_rgb.cpp_ and don't reference the actual board layout - even though the BRD2601B is specifically referenced (as RGB) in the build.gn file - see RGB_LED_ENABLED.

The appropriate include file (eg brd2610b/sl_simple_rgb_pwm_led_rgb_led0_config.h) should be included in the build.gn based on the silabs_board as a dependency (ie - if RGB exists) and pins correctly mapped.

eos1d3, 2023-01-08:
No need to hope Silicon Labs will help there. They are irresponsive. 

Just follow https://developers.home.google.com/matter/vendors/silicon-labs

And I just tested BRD4161A, BRD2703A and your BRD2601B, all built well. No error is found.

I do not have any board right now. Just build all these. So I am considering to get some boards to test later.

grobasoz, 2023-01-08:
You can now build Matter projects in Simplicity Studio.

eos1d3, 2023-01-09:
@grobasoz Did I miss something? I installed and found only 1 Matter example.

grobasoz, 2023-01-09:
I think there are 6 examples, standard lights, switches, locks etc...
What did you install?

eos1d3, 2023-01-09:
It is SimplicityStudio-5 Mac version. After installation, I search "matter" in example and cannot find others.

It did install a lot of libraries. Any additional tools to install?

grobasoz, 2023-01-09:
I haven't tried on the Mac, I'll do so when I get to work this morning... 6:49 am here at the moment...

grobasoz, 2023-01-09:
Matter samples are on Mac as well... eg for BRD4166A (Thunderboard Sense 2)

[image]

[image]

eos1d3, 2023-01-09:
Great thanks!! I will try again.

I tried SimplicityStudio-5 on macOS Monterey earlier without installation issue. I am current installing it on Ventura, it always quit after "Welcome to Simplicity Studio" screen. I tried many times and got the same problem.  Is this a known issue? 

I can switch back to Monterey and test later.

grobasoz, 2023-01-09:
I'm using Monterey - on the [Silabs forum](https://community.silabs.com/s/question/0D58Y00009Zf4vJSAR/when-will-simplicity-studio-5-be-compatible-with-macos-13) others have had issue with Ventura though...

eos1d3, 2023-01-10:
Actually there are a lot more Matter examples for different boards. Using BRD2703A I found 11 examples.

grobasoz, 2023-01-10:
> Actually there are a lot more Matter examples for different boards. Using BRD2703A I found 11 examples.

Thanks for that info! I never really checked boards I don't have here...

Actually I have had to move back to the GitHub release (1.0.2) as the Simplicity versions don't fully work for some reason?
See [here](https://community.silabs.com/s/question/0D58Y00009iPMmySAG/printing-qr-codes?language=en_US)...

Managed to get a lighting-app built and connected to an Apple iPod Mini via Eve/Home.

How's that for "confusing" an already complicated development!?!

silabs-sebastien, 2023-01-10:
Since the original poster is not involved in this issue, the building of this app/board combination is reported as working with the latest SMG and the conversation has moved on to other topics, this issue will now be closed.
If you think you found a bug with using the SMG, please [open a new issue](https://github.com/SiliconLabs/matter/issues/new/choose). If you have any other support requests, please [create a post on our community](https://community.silabs.com/s/topic/0TO1M000000qHZgWAM/matter?language=en_US).

closed 2023-01-10 by silabs-sebastien