SiliconLabs/matter #117: Matter demo over Thread does not work

ShawnHymel, 2023-03-30:
### Description of the issue

I am following along with the tutorial given here: [https://docs.silabs.com/matter/1.0.5/matter-start/](https://docs.silabs.com/matter/1.0.5/matter-start/) but using v1.1.0-1.1 of the software found here. I cannot get the Matter Hub running on my RPi4 to talk to the matter accessory device. Here is my setup:

* Raspberry Pi 4 running the Matter Hub image from [here](https://github.com/SiliconLabs/matter/blob/release_1.1.0-1.1/docs/silabs/general/ARTIFACTS.md#matter-hub-raspberry-pi-image)
* Thunderboard Sense 2 (BRD2601B) running the *ot-rcp-brd2601b.s37* firmware from [here](https://github.com/SiliconLabs/matter/blob/release_1.1.0-1.1/docs/silabs/general/ARTIFACTS.md#radio-co-processor-rcp-images) (without a bootloader) and connected to the Raspberry Pi 4
* Thunderboard Sense 2 (BRD2601B) running the *lighting-app-thread/OpenThread/release/BRD2601B/chip-efr32-lighting-example.s36* application from [here](https://github.com/SiliconLabs/matter/blob/release_1.1.0-1.1/docs/silabs/general/ARTIFACTS.md#matter-accessory-device-images) (with the *bootloader-storage-internal-single-512k-BRD2601B-gsdk4.1.s37* bootloader) and connected to my host PC (for power)

You can see the MAD flashing setup here from Simplicity Studio:

[image]

It appears that my Pi can communicate with the RCP when I give the `ifconfig` command (i.e. wpan0 shows up with an address):

```sh
ubuntu@ubuntu:~$ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.145  netmask 255.255.255.0  broadcast 10.0.0.255
        inet6 fe80::dea6:32ff:febf:835e  prefixlen 64  scopeid 0x20<link>
        inet6 2601:285:8300:7d80:dea6:32ff:febf:835e  prefixlen 64  scopeid 0x0<global>
        inet6 2601:285:8300:7d80::945a  prefixlen 128  scopeid 0x0<global>
        ether dc:a6:32:bf:83:5e  txqueuelen 1000  (Ethernet)
        RX packets 4749  bytes 1026007 (1.0 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1668  bytes 353708 (353.7 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 206  bytes 20992 (20.9 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 206  bytes 20992 (20.9 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wpan0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1280
        inet6 fdde:ad00:beef::ff:fe00:fc00  prefixlen 64  scopeid 0x0<global>
        inet6 fdde:ad00:beef::ff:fe00:fc11  prefixlen 64  scopeid 0x0<global>
        inet6 fdde:ad00:beef:0:8f84:f52b:9505:3c53  prefixlen 64  scopeid 0x0<global>
        inet6 fdde:ad00:beef::ff:fe00:fc10  prefixlen 64  scopeid 0x0<global>
        inet6 fdde:ad00:beef::ff:fe00:fc38  prefixlen 64  scopeid 0x0<global>
        inet6 fd11:22::28d6:5c71:210b:ca23  prefixlen 64  scopeid 0x0<global>
        inet6 fdde:ad00:beef::ff:fe00:a400  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::8c7c:f390:33cd:9fcc  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 16  bytes 2476 (2.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

When I run `mattertool startThread`, that seems to work OK:

```sh
ubuntu@ubuntu:~$ mattertool startThread

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █
 ▀▀    █   █    ▀▀

  ACTIVATOR! This sets your shell environment variables.

[... 125 lines cut ...]
I have posted this issue to the forums, but it seems to be stumping folks there, too: [https://community.silabs.com/s/question/0D58Y00009z7FomSAE/mattertool-matter-hub-with-rcp-cannot-connect-to-endpoint-device-running-light-over-thread-matter-application?language=en_US](https://community.silabs.com/s/question/0D58Y00009z7FomSAE/mattertool-matter-hub-with-rcp-cannot-connect-to-endpoint-device-running-light-over-thread-matter-application?language=en_US).

Any help to get this working would be appreciated!  I have been working on this for weeks, and I am trying to film a video for DigiKey to show off Matter using the SiLabs boards.

### Steps to reproduce the issue

```bash
Please see the steps in the description (as I can add images there)
```

### Device type(s) and network topology

Matter Hub Raspberry Pi connected to OT-RCP on a BRD2601B. Trying to commission and communicate with another BRD2601B running the *lighting-app-thread* MAD application.

### SMG version

release_1.1.0-1.1

### Protocol(s)

Openthread

### Hardware platform(s)

Raspberry Pi 4, 2x Thunderboard Sense 2 (BRD2601B)

ShawnHymel, 2023-03-31:
Based on the recommendation from the forum, I tried flashing the *lighting-app-thread/OpenThread/standard/BRD2601B/chip-efr32-lighting-example.s37* instead of the release version. Still getting the same error:

```
ubuntu@ubuntu:~$ mattertool bleThread
[1680233159.936027][1439:1439] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1680233159.941575][1439:1439] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1680233159.941858][1439:1439] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1680233159.941977][1439:1439] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1680233159.942341][1439:1439] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-FXwBZd)
[1680233160.352151][1439:1439] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1680233160.352322][1439:1439] CHIP:DL: NVS set: chip-counters/reboot-count = 3 (0x3)
[1680233160.353892][1439:1439] CHIP:DL: Got Ethernet interface: eth0
[1680233160.355011][1439:1439] CHIP:DL: Found the primary Ethernet interface:eth0
[1680233160.356049][1439:1439] CHIP:DL: Got WiFi interface: wlan0
[1680233160.356163][1439:1439] CHIP:DL: Failed to reset WiFi statistic counts
[1680233160.356271][1439:1439] CHIP:IN: UDP::Init bind&listen port=0
[1680233160.356451][1439:1439] CHIP:IN: UDP::Init bound to port=36993
[1680233160.356525][1439:1439] CHIP:IN: UDP::Init bind&listen port=0
[1680233160.356677][1439:1439] CHIP:IN: UDP::Init bound to port=49624
[1680233160.356717][1439:1439] CHIP:IN: BLEBase::Init - setting/overriding transport
[1680233160.356753][1439:1439] CHIP:IN: TransportMgr initialized
[1680233160.356800][1439:1439] CHIP:FP: Initializing FabricTable from persistent storage
[1680233160.357015][1439:1439] CHIP:TS: Last Known Good Time: 2023-02-21T10:33:46
[1680233160.359492][1439:1439] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x8FAC4109EE2FA09A, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1680233161.823873][1439:1439] CHIP:ZCL: Using ZAP configuration...
[1680233161.828872][1439:1439] CHIP:DL: MDNS failed to join multicast group on wpan0 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:764: Inet Error 0x00000110: Address not found
[1680233161.829715][1439:1439] CHIP:CTL: System State Initialized...
[1680233161.829848][1439:1439] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1680233161.829981][1439:1439] CHIP:CTL: Setting attestation nonce to random value
[1680233161.830096][1439:1439] CHIP:CTL: Setting CSR nonce to random value
[1680233161.830263][1439:1439] CHIP:IN: UDP::Init bind&listen port=5550
[1680233161.830711][1439:1439] CHIP:IN: UDP::Init bound to port=5550
[1680233161.830805][1439:1439] CHIP:IN: UDP::Init bind&listen port=5550
[1680233161.831039][1439:1439] CHIP:IN: UDP::Init bound to port=5550
[1680233161.831101][1439:1439] CHIP:IN: TransportMgr initialized
[1680233161.831728][1439:1441] CHIP:DL: CHIP task running
[1680233161.832439][1439:1441] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1680233161.832518][1439:1441] CHIP:CTL: Setting attestation nonce to random value
[1680233161.832706][1439:1441] CHIP:CTL: Setting CSR nonce to random value
[1680233161.833707][1439:1441] CHIP:CTL: Generating NOC
[1680233161.834746][1439:1441] CHIP:FP: Validating NOC chain
[1680233161.836762][1439:1441] CHIP:FP: NOC chain validation successful
[1680233161.836949][1439:1441] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1680233161.836987][1439:1441] CHIP:TS: Last Known Good Time: 2023-02-21T10:33:46
[1680233161.837013][1439:1441] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1680233161.837037][1439:1441] CHIP:TS: Retaining current Last Known Good Time
[1680233161.990344][1439:1441] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1680233161.995915][1439:1441] CHIP:TS: Committing Last Known Good Time to storage: 2023-02-21T10:33:46
[1680233162.001498][1439:1441] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 8FAC4109EE2FA09A)
[1680233162.001562][1439:1441] CHIP:IN: UDP::Init bind&listen port=5550
[1680233162.001700][1439:1441] CHIP:IN: UDP::Init bound to port=5550
[1680233162.001733][1439:1441] CHIP:IN: UDP::Init bind&listen port=5550
[1680233162.001835][1439:1441] CHIP:IN: UDP::Init bound to port=5550
[1680233162.001864][1439:1441] CHIP:IN: TransportMgr initialized
[1680233162.031311][1439:1441] CHIP:CTL: Setting thread operational dataset from parameters
[1680233162.031370][1439:1441] CHIP:CTL: Setting attestation nonce to random value
[1680233162.031435][1439:1441] CHIP:CTL: Setting CSR nonce to random value
[1680233162.031483][1439:1441] CHIP:CTL: Commission called for node ID 0x0000000000007CA5
[1680233162.031520][1439:1441] CHIP:DL: Long dispatch time: 200 ms, for event type 2
[1680233162.037227][1439:1440] CHIP:DL: TRACE: Bus acquired for name C-059f
[1680233162.043197][1439:1440] CHIP:DL: FAIL: NULL apEndpoint->mpAdapter in bluezObjectsSetup
[1680233162.043324][1439:1441] CHIP:DL: PlatformBlueZInit init success
[1680233162.043369][1439:1441] CHIP:DIS: Closing all BLE connections
[1680233162.043415][1439:1441] CHIP:BLE: No adapter available for new connection establishment
[1680233281.832379][1439:1439] CHIP:CTL: Shutting down the commissioner
[... cut ...]
[1680233281.837621][1439:1439] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1680233281.837686][1439:1439] CHIP:DL: Inet Layer shutdown
[1680233281.837740][1439:1439] CHIP:DL: BLE shutdown

(process:1439): GLib-GObject-CRITICAL **: 03:28:01.837: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
[1680233281.838616][1439:1439] CHIP:DL: System Layer shutdown
[1680233281.839288][1439:1439] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:512: CHIP Error 0x00000032: Timeout
The Node id of the commissioned device is 31909
```

silabs-sebastien, 2023-04-05:
Hi @ShawnHymel,
As you mentioned, you already have a [thread on the community](https://community.silabs.com/s/question/0D58Y00009z7FomSAE/mattertool-matter-hub-with-rcp-cannot-connect-to-endpoint-device-running-light-over-thread-matter-application?language=en_US) where you are receiving support from Silabs. Please continue discussing this issue there as this does not appear to be a bug at this time.

closed 2023-04-05 by silabs-sebastien