SiliconLabs/matter #133: Generated .xcframework used in sample application fails in commissioning process

rahil-shrimali-ei, 2023-06-10:
### General description of the feature

- My purpose is to commission matter enabled Wi-Fi smart plug from sample application.
- I am successfully able to perform commissioning operation using CHIPTOOL app **(src/darwin/CHIPTool)**. 
- I have generated Matter.xcframework by applying following commands on path: **src/darwin/Framework**

`xcodebuild clean archive \
  -scheme 'Matter Framework' \
-configuration Release \
-destination 'generic/platform=iOS' \
-archivePath './build/Matter.framework-iphoneos.xcarchive' \
  BUILD_LIBRARY_FOR_DISTRIBUTION=YES \
  SKIP_INSTALL=NO`
  
  `xcodebuild clean archive \
  -scheme 'Matter Framework' \
-configuration Release \
-destination 'generic/platform=iOS Simulator' \
-archivePath './build/Matter.framework-iOS-Simulator.xcarchive' \
  BUILD_LIBRARY_FOR_DISTRIBUTION=YES \
  ONLY_ACTIVE_ARCH=NO \
  EXCLUDED_ARCHS=arm64 \
  SKIP_INSTALL=NO`
  
  `xcodebuild -create-xcframework \
-framework './build/Matter.framework-iOS-Simulator.xcarchive/Products/Library/Frameworks/Matter.framework' \
-framework './build/Matter.framework-iphoneos.xcarchive/Products/Library/Frameworks/Matter.framework' \
-output './build/Matter.xcframework'`

It looks like generated framework is not able to access the file from other folders like **(src/app)** , **(src/messaging)**.

Can anyone help me here to generate the Matter.xcframework which contains every useful resource?

### SMG version

release_1.1.0-1.1

### Protocol(s)

Wi-Fi, Bluetooth

### Hardware platform(s)

Mini Smart Wi-Fi Plug

silabs-sebastien, 2023-06-13:
@rahil-shrimali-ei I'm looking into your report and I will get back to you shortly.

silabs-sebastien, 2023-06-19:
@rahil-shrimali-ei This is just a quick update to let you know it's taking some time to investigate this issue. In the meantime, did you try anything else or have any new progress on your end?

Also, did you try this process with the newest SMG release (2.0.0-1.1)?

closed 2023-08-17 by silabs-sebastien