SiliconLabs/matter #134: [BUG] Thread Network Scan leads to HEAP-Fault when no Thread-Networks are in Range

ldoering, 2023-07-05:
### Description of the issue

During commissioning the Alexa App is sending a Network Scan Command via BLE.
When no Network is in range the vApplicationMallocFailedHook is triggered and the device will freeze. As soon as at least one Network is in Range the problem does not occur.

In network-commissioning.cpp's Instance::OnFinished (Line 545) "scanResponseArray.Alloc" for a count of "0" seems to trigger this.

The Issues does not seem to be triggered by other ecosystems, since they do not use this ScanNetwork function

### Steps to reproduce the issue

```bash
1. Factory Reset a Thread Device
2. Disable all Thread Networks in Range
3. Either one of the following:
3a. Start Comissioning using a Matter-but-not-Thread-capable Echo device and the Android Alexa app, which issues a Scan Request during commissioning.
3b. I assume it would also be possible to run the ScanNetwork Command from the Network Commissioning Cluster using CHiptool via BLE.
4. Observe the Assert in the RTT-Log
```

### Device type(s) and network topology

Tested using a EFR32MG24 Device, a Echo Dot Gen 2 and a Android Phone.

### Proposed solution(s)

Filter the Case (networks->Count() == 0) in network-commissioning.cpp

### SMG version

release_2.0.0-1.1

### Protocol(s)

Thread, Bluetooth

### Hardware platform(s)

EFR32MG24

silabs-sebastien, 2023-07-07:
@ldoering Thanks for the report! We're looking into the issue and will work on a fix shortly. I'll post an update here once I have more info.

silabs-sebastien, 2023-07-11:
@ldoering A fix has been proposed by our team. You can see the PR over [here](https://github.com/project-chip/connectedhomeip/pull/27896).

silabs-sebastien, 2023-07-11:
I'll close this issue for now as a resolution is in the process of being applied to the CSA public repo already.