SiliconLabs/matter #137: bash: ./scripts/examples/gn_efr32_example.sh: No such file or directory

ifdreams, 2023-11-02:
### Description of the issue

I can't compile the project through gn_efr32_example.sh anymore.

### Steps to reproduce the issue

```bash
./scripts/examples/gn_efr32_example.sh  ./examples/lighting-app/silabs/efr32/ ./out/lighting-app BRD4186C
```

### Device type(s) and network topology

Matter over Thread

### SMG version

release_2.2.0-1.2

### Protocol(s)

Thread

### Hardware platform(s)

BRD4186C

### Development platform(s)

MG24

### Captured data or logs

matter$ ./scripts/examples/gn_efr32_example.sh  ./examples/lighting-app/silabs/efr32/ ./out/lighting-app BRD4186C
bash: ./scripts/examples/gn_efr32_example.sh: No such file or directory

ifdreams, 2023-11-02:
gn_efr32_example.sh change gn_silabs_example.sh

silabs-sebastien, 2023-11-21:
> gn_efr32_example.sh change gn_silabs_example.sh

Yes, instead you should be using something like:
`./scripts/examples/gn_silabs_example.sh ./examples/lighting-app/silabs ./out/light-switch-app_ICD BRD4186C`
and add  `--icd` if you want an intermittently connected device.

closed 2023-11-21 by silabs-sebastien