SiliconLabs/matter #139: Can't find the BRD2704A bootloader in bootloader_binaries_2.2.0-1.2.zip.

lboue, 2024-01-11:
### Description of the issue

I can't find the BRD2704A / Sparkfun Thing Plus MGM240P bootloader in bootloader_binaries_2.2.0-1.2.zip.

[image]

### Steps to reproduce the issue

```bash
1) Download https://github.com/SiliconLabs/matter/releases/download/v2.2.0-1.2/bootloader_binaries_2.2.0-1.2.zip
2) Unzip bootloader_binaries_2.2.0-1.2.zip
3) Go to bootloader_binaries_2.2.0-1.2 directory

BRD2704A / Sparkfun Thing Plus MGM240P
```

### Device type(s) and network topology

BRD2704A / Sparkfun Thing Plus MGM240P board

### SMG version

release_2.2.0-1.2

### Protocol(s)

Thread

### Hardware platform(s)

BRD2704A

lboue, 2024-01-11:
Same issue with :
 * [bootloader_binaries_1.1.0-1.1.zip](https://github.com/SiliconLabs/matter/releases/download/v1.1.0-1.1/bootloader_binaries_1.1.0-1.1.zip)
 * [bootloader_binaries_2.1.0-1.1.zip](https://github.com/SiliconLabs/matter/releases/download/v2.1.0-1.1/bootloader_binaries_2.1.0-1.1.zip)

jmartinez-silabs, 2024-01-17:
fwiw, you can find it here
https://github.com/SiliconLabs/sdk_support/tree/main/matter/efr32/bootloader_binaries

But we should add it to the artifacts also.

lboue, 2024-01-17:
> But we should add it to the artifacts also.

Feel free to post a message here when this is corrected.

silabs-sebastien, 2024-03-26:
@lboue The binaries are now available in the release files as expected for 2.3.0-1.3-alpha.2 and beta.1.

closed 2024-03-26 by silabs-sebastien