SiliconLabs/matter #140: ImportError: cannot import name 'OneStyleAndTextTuple' from 'prompt_toolkit.formatted_text'

lboue, 2024-02-11:
### Description of the issue

Setting environment variables for Host tools fails : 

```
+++ echo -n '  Setting environment variables for Host tools.............'
  Setting environment variables for Host tools.............++++ echo /home/user/matter/.environment/pigweed-venv/bin:/home/user/matter/.environment/cipd/packages/arm/bin:/home/user/matter/.environment/cipd/packages/arm:/home/user/matter/.environment/cipd/packages/pigweed/bin:/home/user/matter/.environment/cipd/packages/pigweed:/home/user/matter/.environment/cipd/packages/zap:/home/user/matter/.environment/cipd:/home/user/matter/third_party/pigweed/repo/out/host/host_tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++++ sed 's|:/home/user/matter/third_party/pigweed/repo/out/host/host_tools$||g;'
++++ sed 's|^/home/user/matter/third_party/pigweed/repo/out/host/host_tools:||g;'
++++ sed 's|:/home/user/matter/third_party/pigweed/repo/out/host/host_tools:|:|g;'
+++ PATH=/home/user/matter/.environment/pigweed-venv/bin:/home/user/matter/.environment/cipd/packages/arm/bin:/home/user/matter/.environment/cipd/packages/arm:/home/user/matter/.environment/cipd/packages/pigweed/bin:/home/user/matter/.environment/cipd/packages/pigweed:/home/user/matter/.environment/cipd/packages/zap:/home/user/matter/.environment/cipd:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+++ export PATH
+++ PATH=/home/user/matter/third_party/pigweed/repo/out/host/host_tools:/home/user/matter/.environment/pigweed-venv/bin:/home/user/matter/.environment/cipd/packages/arm/bin:/home/user/matter/.environment/cipd/packages/arm:/home/user/matter/.environment/cipd/packages/pigweed/bin:/home/user/matter/.environment/cipd/packages/pigweed:/home/user/matter/.environment/cipd/packages/zap:/home/user/matter/.environment/cipd:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+++ export PATH
+++ '[' -z '' ']'
+++ echo done
done
+++ '[' -z '' ']'
+++ echo ''

+++ '[' -n /usr/bin/bash -o -n '' ']'
+++ hash -r
+++ '[' -z '' ']'
+++ echo 'Checking the environment:'
Checking the environment:
+++ '[' -z '' ']'
+++ echo ''

+++ '[' -z '' ']'
+++ pw --no-banner --loglevel info doctor
Traceback (most recent call last):
  File "/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/plugins.py", line 96, in from_name
    module = importlib.import_module(module_name)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/pw_watch/watch.py", line 93, in <module>
    import pw_console.python_logging
  File "/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/pw_console/__init__.py", line 16, in <module>
    from pw_console.embed import PwConsoleEmbed
  File "/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/pw_console/embed.py", line 23, in <module>
    from pw_console.console_app import ConsoleApp
  File "/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/pw_console/console_app.py", line 59, in <module>
    from ptpython.layout import CompletionVisualisation  # type: ignore
  File "/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/ptpython/__init__.py", line 3, in <module>
    from .repl import embed
  File "/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/ptpython/repl.py", line 23, in <module>
    from prompt_toolkit.formatted_text import OneStyleAndTextTuple
ImportError: cannot import name 'OneStyleAndTextTuple' from 'prompt_toolkit.formatted_text' (/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/prompt_toolkit/formatted_text/__init__.py)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/user/matter/.environment/pigweed-venv/bin/pw", line 8, in <module>
    sys.exit(main())
  File "/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/__main__.py", line 43, in main
    pw_command_plugins.register()
  File "/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/pw_command_plugins.py", line 64, in register
    _register_builtin_plugins(plugin_registry)
  File "/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/pw_command_plugins.py", line 42, in _register_builtin_plugins
    registry.register_by_name('watch', 'pw_watch.watch', 'main')
  File "/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/plugins.py", line 303, in register_by_name
    Plugin.from_name(name, module_name, member_name, source)
  File "/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/pw_cli/plugins.py", line 104, in from_name
    raise Error(f'Failed to import module "{module_name}"') from err
pw_cli.plugins.Error: Failed to import module "pw_watch.watch" (ImportError: cannot import name 'OneStyleAndTextTuple' from 'prompt_toolkit.formatted_text' (/home/user/matter/.environment/pigweed-venv/lib/python3.10/site-packages/prompt_toolkit/formatted_text/__init__.py))
```

### Steps to reproduce the issue

```bash
https://github.com/SiliconLabs/matter
	cd matter
	./scripts/examples/gn_silabs_example.sh ./silabs_examples/dishwasher-app/silabs/ ./out/dishwasher-app BRD2704A
```

### Device type(s) and network topology

Setup up build environnement 

### SMG version

release_2.2.0-1.2

### Protocol(s)

Thread

### Hardware platform(s)

BRD2704A 

### Development platform(s)

Ubuntu 22.04.3 LTS (GNU/Linux 6.5.0-17-generic x86_64)

shuiYe-704265085, 2024-02-27:
My situation differs from yours; I encounter this error when running the environment update command "source ./scripts/activate.sh". You might also resolve it by changing the version of prompt-toolkit, which is located in the file scripts/setup/constraints.txt. Search for prompt-toolkit and change the version to 3.0.43.

lboue, 2024-03-02:
> scripts/setup/constraints.txt

Thanks, that fixed the problem I was having.
I open PR #147 to fix it.