SiliconLabs/matter #145: [dishwasher] LED is not blinking when pausing dishwasher

lboue, 2024-02-17:
### Description of the issue

LED is not blinking when pausing dishwasher from matter-accessory-device-images_2.2.0-1.2/release/BRD2703A/OpenThread/**dishwasher-app-no-lcd-thread.s37**

```
00> [00:04:38.182][silabs ]Pausing dishwasher
00> [00:04:38.182][detail][DMG] Endpoint 1, Cluster 0x0000_0060 update version to 358ba048
00> [00:04:38.182][detail][DMG] Cannot merge the new path into any existing path, create one.
00> [00:04:38.182][info  ][ZCL] Cluster callback: 0x0000_0060
00> [00:04:38.182][detail][DMG] Building Reports for ReadHandler with LastReportGeneration = lu DirtyGeneration = lu
00> [00:04:38.187][detail][DMG] <RE:Run> Cluster 60, Attribute 4 is dirty
00> [00:04:38.187][detail][DMG] Reading attribute: Cluster=0x0000_0060 Endpoint=1 AttributeId=0x0000_0004 (expanded=1)
00> [00:04:38.187][error ][ZCL] OperationalState: Reading
00> [00:04:38.187][error ][ZCL] OperationalState: H1
00> [00:04:38.189][detail][DMG] Fetched 0 events
00> [00:04:38.189][detail][DMG] <RE> Sending report (payload has 40 bytes)...
00> [00:04:38.190][info  ][EM] <<< [E:63657i S:595 M:187742777] (S) Msg TX to 1:00000000212740BC [FE0B] [UDP:[fda9:150:8b20:0:2194:5663:f84e:513d%ot]:62942] --- Type 0001:05 (IM:ReportData)
00> [00:04:38.191][detail][DMG] IM RH moving to [AwaitingReportResponse]
00> [00:04:38.192][detail][DMG] <RE> ReportsInFlight = 1 with readHandler 0, RE has no more messages
00> [00:04:38.192][detail][DMG] All ReadHandler-s are clean, clear GlobalDirtySet
00> [00:04:38.220][detail][IN] UDP Message Received packet nb : 100 SrcAddr : fda9:150:8b20:0:2194:5663:f84e:513d[62942] DestAddr : fda9:150:8b20:0:7ec9:223c:a6eb:fb9c[5540] Payload Length 42
00> [00:04:38.221][info  ][EM] >>> [E:63657i S:595 M:217972059 (Ack:187742777)] (S) Msg RX from 1:00000000212740BC [FE0B] --- Type 0001:01 (IM:StatusResponse)
00> [00:04:38.221][detail][EM] Found matching exchange: 63657i, Delegate: 0x2000a664
00> [00:04:38.221][detail][EM] Rxd Ack; Removing MessageCounter:187742777 from Retrans Table on exchange 63657i
00> [00:04:38.221][info  ][IM] Received status response, status is 0x00
00> [00:04:38.221][detail][DMG] <RE> OnReportConfirm: NumReports = 0
00> [00:04:38.221][detail][DMG] IM RH moving to [CanStartReporting]
00> [00:04:38.222][info  ][EM] <<< [E:63657i S:595 M:187742778 (Ack:217972059)] (S) Msg TX to 1:00000000212740BC [FE0B] [UDP:[fda9:150:8b20:0:2194:5663:f84e:513d%ot]:62942] --- Type 0000:10 (SecureChannel:StandaloneAck)
00> [00:04:38.223][detail][EM] Flushed pending ack for MessageCounter:2179
```

### Steps to reproduce the issue

```bash
1. Flash matter-accessory-device-images_2.2.0-1.2/release/BRD2703A/OpenThread/dishwasher-app-no-lcd-thread.s37 to BRD2703A board
2. Start dishwasher
3. Pause dishwasher
```

### Device type(s) and network topology

Raspberry Pi 4

### SMG version

release_2.2.0-1.2

### Protocol(s)

Thread

### Hardware platform(s)

EFR32xG24 Explorer Kit BRD2703A

lboue, 2024-03-02:
Waiting for #149 approval.

silabs-sebastien, 2024-03-26:
@lboue I will look at your PR shortly to integrate it into an upcoming release (most likely April, but still TBD). I'll update here again once it has been processed for a future release. Thanks for the info on that bug, btw! Much appreciated! 👍

closed 2024-11-03 by lboue