SiliconLabs/matter #146: Ecosystems validation suggestions

hslin224, 2024-02-27:
### General description of the feature

1. For Samsung Hub, Smart Things WIFI/HUB (Aeotec) is not equipped with Thread Radio. Suggest to use Samsung SmartThings Station for evaluation usage.
2. The Amazon Echo Dot is not equipped with Thread Radio. Suggest to use Amazon Echo(4th gen) for evaluation usage.

### SMG version

release_2.2.0-1.2

### Protocol(s)

Thread

### Hardware platform(s)

MG21, MG24

lboue, 2024-03-18:
@hslin224 
Check [Silicon Labs Matter GitHub v2.3.0-1.3-beta.1](https://github.com/SiliconLabs/matter/releases/tag/v2.3.0-1.3-beta.1) release.
Samsung Hub has been added
- Hardware: Smart Things Station
- Thread: Software: 1.8.12.21
- Wi-Fi: Software: 2.3.29-7

hslin224, 2024-03-19:
@lboue 
Great! Do you know if there're documents for test scenarios?

lboue, 2024-03-21:
> @lboue Great! Do you know if there're documents for test scenarios?

You should ask SiliconLabs for that.

rerasool, 2024-03-26:
Good catch. I checked with the team and the hub details used for Thread testing were missed in the release notes. This has been addressed in the latest release as mentioned by @lboue : https://github.com/SiliconLabs/matter/releases/tag/v2.3.0-1.3-beta.1

Test scenarios were also shared with @hslin224 separately.

Closing the issue, please reopen if needed, thanks.

closed 2024-03-26 by rerasool