SiliconLabs/matter #152: Error when setting up Python environment

vuhailongkl97, 2024-04-12:
### Description of the issue

when run scripts/activate.sh i got this error

### Steps to reproduce the issue

```bash
run source ./scripts/activate.sh
```

### Device type(s) and network topology

linux os

### SMG version

release_2.2.0-1.2

### Protocol(s)

Wi-Fi

### Hardware platform(s)

mg24

### Development platform(s)

linux

### Captured data or logs

```
tter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
FAILED: python/gen/matter_build_venv/compiled_requirements.txt
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/linux:linux_x64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
  ERROR: Cannot install -r python/gen/matter_build_venv/generated_requirements.txt (line 84), protobuf and protobuf~=3.20.1 because these package versions have conflicting dependencies.
Traceback (most recent call last):
  File "/home/build-user/iot/matter/.environment/gn_out/python-venv/lib/python3.10/site-packages/pip/_vendor/resolvelib/resolvers.py", line 316, in _backjump
    name, candidate = broken_state.mapping.popitem()
KeyError: 'dictionary is empty'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/build-user/iot/matter/.environment/gn_out/python-venv/lib/python3.10/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 92, in resolve
    result = self._result = resolver.resolve(
  File "/home/build-user/iot/matter/.environment/gn_out/python-venv/lib/python3.10/site-packages/pip/_vendor/resolvelib/resolvers.py", line 546, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
  File "/home/build-user/iot/matter/.environment/gn_out/python-venv/lib/python3.10/site-packages/pip/_vendor/resolvelib/resolvers.py", line 434, in resolve
    success = self._backjump(causes)
  File "/home/build-user/iot/matter/.environment/gn_out/python-venv/lib/python3.10/site-packages/pip/_vendor/resolvelib/resolvers.py", line 318, in _backjump
    raise ResolutionImpossible(causes)
pip._vendor.resolvelib.resolvers.ResolutionImpossible: [RequirementInformation(requirement=SpecifierRequirement('protobuf'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf~=3.20.1'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0.dev0,>=3.19.5'), parent=LinkCandidate('https://files.pythonhosted.org/packages/dc/a6/12a0c976140511d8bc8a16ad15793b2aef29ac927baa0786ccb7ddbb6e1c/googleapis_common_protos-1.63.0-py2.py3-none-any.whl (from https://pypi.org/simple/googleapis-common-protos/) (requires-python:>=3.7)')), RequirementInformation(requirement=SpecifierRequirement('protobuf'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf~=3.20.1'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0.dev0,>=3.19.5'), parent=LinkCandidate('https://files.pythonhosted.org/packages/f0/43/c9d8f75ddf08e2a0a27db243c13a700c3cc7ec615b545b697cf6f715ad92/googleapis_common_protos-1.62.0-py2.py3-none-any.whl (from https://pypi.org/simple/googleapis-common-protos/) (requires-python:>=3.7)')), RequirementInformation(requirement=SpecifierRequirement('protobuf'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf~=3.20.1'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0.dev0,>=3.19.5'), parent=LinkCandidate('https://files.pythonhosted.org/packages/21/49/12996dc0238e017504dceea1d121a48bd49fb3f4416f40d59fc3e924b4f3/googleapis_common_protos-1.61.0-py2.py3-none-any.whl (from https://pypi.org/simple/googleapis-common-protos/) (requires-python:>=3.7)')), RequirementInformation(requirement=SpecifierRequirement('protobuf'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf~=3.20.1'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0.dev0,>=3.19.5'), parent=LinkCandidate('https://files.pythonhosted.org/packages/a7/bc/416a1ffeba4dcd072bc10523dac9ed97f2e7fc4b760580e2bdbdc1e2afdd/googleapis_common_protos-1.60.0-py2.py3-none-any.whl (from https://pypi.org/simple/googleapis-common-protos/) (requires-python:>=3.7)')), RequirementInformation(requirement=SpecifierRequirement('protobuf'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf~=3.20.1'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0.dev0,>=3.19.5'), parent=LinkCandidate('https://files.pythonhosted.org/packages/b3/b7/bbaa556e9ff0580f408c64ccf4db0c1414eec79e7151d33a10bc209ffb6d/googleapis_common_protos-1.59.1-py2.py3-none-any.whl (from https://pypi.org/simple/googleapis-common-protos/) (requires-python:>=3.7)')), RequirementInformation(requirement=SpecifierRequirement('protobuf'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf~=3.20.1'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0dev,>=3.19.5'), parent=LinkCandidate('https://files.pythonhosted.org/packages/a7/8d/7ccf8f63ab0a768e13720262374c0d6a1489f7f23f1b89a16c92af452f09/googleapis_common_protos-1.59.0-py2.py3-none-any.whl (from https://pypi.org/simple/googleapis-common-protos/) (requires-python:>=3.7)')), RequirementInformation(requirement=SpecifierRequirement('protobuf'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf~=3.20.1'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0dev,>=3.19.5'), parent=LinkCandidate('https://files.pythonhosted.org/packages/32/4e/ed585842aaa704d87495a0e99317aaa44c5007a597c05b995fa8cfc4dfbe/googleapis_common_protos-1.58.0-py2.py3-none-any.whl (from https://pypi.org/simple/googleapis-common-protos/) (requires-python:>=3.7)')), RequirementInformation(requirement=SpecifierRequirement('protobuf'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf~=3.20.1'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0dev,>=3.19.5'), parent=LinkCandidate('https://files.pythonhosted.org/packages/76/b9/fc6a31a90e3c78cea00d209d48f1e5c6c9edc6b14f3039470c3f7fae60a1/googleapis_common_protos-1.57.1-py2.py3-none-any.whl (from https://pypi.org/simple/googleapis-common-protos/) (requires-python:>=3.7)')), RequirementInformation(requirement=SpecifierRequirement('protobuf'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf~=3.20.1'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0dev,>=3.19.5'), parent=LinkCandidate('https://files.pythonhosted.org/packages/f0/2a/25d8c1ceedc5af97de37434c9c5e38ce28aaa45960aa2bd7aa215fc420c0/googleapis_common_protos-1.57.0-py2.py3-none-any.whl (from https://pypi.org/simple/googleapis-common-protos/) (requires-python:>=3.7)'))]

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
[... cut ...]
  File "/home/build-user/iot/matter/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 433, in install
    install_packages(gn_target)
  File "/home/build-user/iot/matter/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 414, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['ninja', '-C', '/home/build-user/iot/matter/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.
Installing pip requirements for all...

```

vuhailongkl97, 2024-04-12:
i use this os 
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

gulbeta, 2024-05-03:
I have the exact same issue.

lboue, 2024-05-03:
This is an Python `protobuf `library issue:
```
pip._vendor.resolvelib.resolvers.ResolutionImpossible: [RequirementInformation(requirement=SpecifierRequirement('protobuf'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf~=3.20.1'), parent=None), RequirementInformation(requirement=SpecifierRequirement('protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0.dev0,>=3.19.5'), parent=LinkCandidate('https://files.pythonhosted.org/packages/dc/a6/12a0c976140511d8bc8a16ad15793b2aef29ac927baa0786ccb7ddbb6e1c/googleapis_common_protos-1.63.0-py2.py3-none-any.whl (from https://pypi.org/simple/googleapis-common-protos/) (requires-python:>=3.7)')), 
```

lboue, 2024-05-03:
It looks like this issue: [[Build] Bootstraph.sh fails with pip protobuf error #32533 ](https://github.com/project-chip/connectedhomeip/issues/32533) in connectedhomeip Github repo.